bmpenuelas/systemverilog-formatter-vscode #14: SystemVerilog Formatter - Format this file failed

ARC-MX, 2022-05-08:
[image]

Reset816, 2022-05-09:
got same error

Reset816, 2022-05-09:
> got same error

on 2.16.0 and 2.16.1

ARC-MX, 2022-05-09:
> 

所以你现在也是不能用吗

Reset816, 2022-05-09:
> 

yes

bmpenuelas, 2022-05-12:
Hi, I have just published v2.16.2 to fix a problem with some paths that need to be quoted. Can you try this one and let me know if it still fails? Perhaps uninstall and install from scratch.

Reset816, 2022-05-13:
> Hi, I have just published v2.16.2 to fix a problem with some paths that need to be quoted. Can you try this one and let me know if it still fails? Perhaps uninstall and install from scratch.

I'm still getting same error with v2.16.2
[image]

Reset816, 2022-05-13:
> > Hi, I have just published v2.16.2 to fix a problem with some paths that need to be quoted. Can you try this one and let me know if it still fails? Perhaps uninstall and install from scratch.
> 
> I'm still getting same error with v2.16.2 [image]

VSCode 1.67.1 and Windows 11 22000.652. And the paths do not contain spaces, which are all English characters

Reset816, 2022-05-15:
> Hi, I have just published v2.16.2 to fix a problem with some paths that need to be quoted. Can you try this one and let me know if it still fails? Perhaps uninstall and install from scratch.

Is there any way to get log and provide more information for you?

bmpenuelas, 2022-06-03:
Hello, I have been trying to figure out what could be failing for you guys, since it works all right for most people. 

I've published v2.16.3, can you please try and see if it is solved with this patch? (maybe try a clean install). If it still fails please let me know. 

(in case of failure with the latest version, you can click `Help > Toggle Developer Tools`, go to the `Console` tab, and see if you get any error messages there when it fails)

Reset816, 2022-06-04:
> Hello, I have been trying to figure out what could be failing for you guys, since it works all right for most people.
>
> I've published v2.16.3, can you please try and see if it is solved with this [...]

It works great now.
Thank you.

bmpenuelas, 2022-06-08:
Happy to hear that! Do not hesitate to let me know if you encounter any other issues in the future!

closed 2022-06-08 by bmpenuelas