bmpenuelas/systemverilog-formatter-vscode #15: Enhancement: Keyboard Shortcut for running document format

Tekunalogy, 2022-06-01:
Since the extension doesn't have default keybinds, I experimented and used these keyboard shortcuts. These mimic the same functionality that vscode provides for supported languages.

For supported languages, `ctrl + k, ctrl + f` autoformats the selection
For supported languages, `alt + shift + f` autoformats the entire document

```
[
    {
        "key": "ctrl+k ctrl+f",
        "command": "systemverilog-formatter-vscode.formatSelection",
        "when": "editorTextFocus && !editorReadonly && resourceExtname == '.sv'"
    },
    {
        "key": "shift+alt+f",
        "command": "systemverilog-formatter-vscode.formatDocument",
        "when": "editorTextFocus && !editorHasDocumentFormattingProvider && !editorReadonly && resourceExtname == '.sv'"
    }
]
```

I added a check so that it only works for `.sv` filetypes only (will not interfere with other filetypes or other shortcuts). I tried to find a way to add this within the extension's source code, but I haven't spent enough time with it to figure it out.

Hopefully this can be implemented in the next update!

bmpenuelas, 2022-06-01:
The extension is integrated with the built-in `Format Document` command, so the shortcut should work out of the box. 

I have just tried it with the latest version and `shift+alt+f` worked without adding additional settings (you must be on a verilog or sv file, and the right language should be selected in the bottom bar). You can also try to trigger the "native" `Format Document` command by pressing `ctrl+shift+p` and then Format Document (selecting the one that does not belong to the extension).

[image]

Tekunalogy, 2022-06-03:
Oh interesting. It didn't work for me until I updated the extension just now. But that is good to know. Thank you!

closed 2022-06-03 by Tekunalogy