bmpenuelas/systemverilog-formatter-vscode #16: 2.16.3 doesn't work on macos

pinkcatfly, 2022-06-07:
I notice this doesn't officially support MACOS (only for now, I wish)...BUT I really need to it so I decided to give it a try anyway.
I use brew to install verible, and I can get the "verible-verilog-format" run in the terminal.
So I try to use 2.16.3 with {"systemverilogFormatter.veribleBuild": "none"} in the setting SJON. The log I got is 

#命令"SystemVerilog Formatter - Format this file"导致错误 (Command failed:   -
/bin/sh: -: command not found
)

After roll back to 2.16.2 everything seems to work just fine.......

bmpenuelas, 2022-06-08:
Hi! I have just published 2.16.4 adressing this issue. Can you please try it and let me know if it worked?

pinkcatfly, 2022-06-13:
Nice ！seems 2.16.2 works！I have a little personal problem. I want each "begin" wraps to next line, just like what "{" do in GNU C formatter.  
I think i check ever verible-verilog-format's command line arguments but none seems to do the work for me. Maybe you know what to use?
By the way, "verible-verilog-format" install from brew is an old versioned with too few command arguments. So I clone it and compile the latest version, which I think has all arguments. THX!

bmpenuelas, 2022-06-13:
You mentioned 2.16.2 works before, have you also tried the latest release 2.16.4?

(for command line arguments of verible you can add them to this extension with `systemverilogFormatter.commandLineArguments`, but I only know about the arguments documented in https://chipsalliance.github.io/verible/verilog_format.html , for more commands you can perhaps request it in their project page)

pinkcatfly, 2022-06-24:
My mistake .
"Nice ！seems 2.16.2 works！" is actually talking about 2.16.4.
I have use 2.16.4 for a while. The basic function works all fine. Only one thing I think is abnormal here.
When I changing 'systemverilogFormatter.commandLineArguments' in VScode Setting GUI (not JSON), the modify seem not to effect immediately but only after I close and restart VScode. Maybe the setting is checked only when starting up and other plugin, which take effect immediately, may have a way working around this.

pinkcatfly, 2022-06-24:
Maybe u can leave it be since vscode is a swift editor and spend a second to restart wont be a problem. Thanks for developing the best formatter plugin for verilog~~~