bmpenuelas/systemverilog-formatter-vscode #17: TypeError: Cannot read properties of undefined (reading 'body')

ARC-MX, 2022-06-17:
我在一个新机器上安装这个插件，是安装在vscode远程宿主机上的，发现一运行就报这个错误不知道是为什么
单独下载verible 使用verible-verilog-format是正常的。
`SystemVerilog: Indexing: Unable to process file: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
TypeError: Cannot read properties of undefined (reading 'body')
Auto Indexing: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
SystemVerilog: Indexing: Unable to process file: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
TypeError: Cannot read properties of undefined (reading 'body')
Auto Indexing: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
SystemVerilog: Indexing: Unable to process file: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
TypeError: Cannot read properties of undefined (reading 'body')
Auto Indexing: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
SystemVerilog: Indexing: Unable to process file: file:///data/mxwang/Project/FPGA/ZCU104/wpu_acc_mig_base/wpu_acc_mig_base.srcs/sim_1/imports/new/tb_axi_mig_wrapper.sv
TypeError: Cannot read properties of undefined (reading 'body')`

bmpenuelas, 2022-06-18:
Hi, can you please describe your issue in English?

Also please attach the logs form the

bmpenuelas, 2024-06-25:
Closing since no more info was provided, feel free to reopen it if you can provide more info to reproduce the issue and logs.

closed 2024-06-25 by bmpenuelas