bmpenuelas/systemverilog-formatter-vscode #19: Incorrect shell used

civec, 2023-06-07:
I have Verible binaries in the PATH and they are accessible in my BASH terminal. VS Code is trying to run Verible with /bin/sh even though BASH is set in VS Code terminal.integrated.shell.linux

`2023-06-07 12:16:02.708 [error] [bmpenuelas.systemverilog-formatter-vscode] provider FAILED`
`2023-06-07 12:16:02.708 [error] Error: Command failed: verible-verilog-format  -`
`/bin/sh: verible-verilog-format: command not found`

Could you please help me with this issue?

bmpenuelas, 2023-06-08:
Of course! I have created a new version that uses terminal.integrated.shell (if present) to set the shell to be used. 

I have just published it as pre-release. You need to find the extension inside VSCode, click the drop-down and select it (for more info see https://github.com/microsoft/vscode-docs/blob/vnext/release-notes/v1_63.md#pre-release-extensions )

Can you please try it and let me know how it goes?

the-moog, 2023-09-28:
Similar issue here (using the prerelease)  despite setting the path to 'none', it appears to be still trying to use the packaged binary (and then for the wrong os/kernel) (note: running external binaries is unacceptable here)  I've put a build from official source version is in my local user bin folder which is in the path and it does not find it.  The debug logging is a little light so I can't say more than 'permission denied'.

bmpenuelas, 2025-04-25:
Hi, maybe what you encountered was due to the fact that previously the extension was updating its settings only on the first activation. So maybe you were not seeing your changes immediately when you switched to 'none'.  I've just published an update that takes in the config changes immediately, and logs more error details, so debugging should be easier.

closed 2025-04-25 by bmpenuelas