bmpenuelas/systemverilog-formatter-vscode #22: system error (spawnSync /bin/sh EPIPE)

david-shmailov, 2024-06-16:
I've installed the plugin, using CentOS 7 in the configuration as this is my OS, whenever i try to format a file or selection i get this error:
system error (spawnSync /bin/sh EPIPE)

bmpenuelas, 2024-06-25:
That looks like the spawned process either terminated unexpectedly, or there was too much data and the buffer was filled. Does this happen with small files too? You could try using the "none" setting and providing a path to some other binary just to see if you get a different message.

david-shmailov, 2024-06-25:
you are correct.
I also had to add executable permission to the binary of CentOS7 formatter in the plugin directory, and after i did, it works without displaying an error. but on large file it doesn't change anything, and doesn't display an error anymore.
on small files it seems to work.
anyway i can get better log reports from vscode on whats going on?

closed 2024-06-25 by david-shmailov

bmpenuelas, 2024-06-25:
Happy to hear that, thanks for the update!