bmpenuelas/systemverilog-formatter-vscode #28: Generates endless newlines when an error occurs or after a comment

CatalpaEel, 2025-05-18:
Hello! Your formatter has been extremely helpful to me, but I seem to have found a small bug. 

## Description

When I add a single-line comment in my Verilog code and then format it, an empty line is generated after the comment. This happens every time I format the code, resulting in a large number of unnecessary blank lines over time.

Also, if I write a syntax error in my Verilog code and then format it, empty lines will be generated after all the original lines.

## Examples

### After a comment

Before formatting:
```
module top_module (
    input clk,
    input areset,
    input [7:0] d,
    output reg [7:0] q // Here is a comment
);
  always @(posedge clk or posedge areset) begin
    if (areset == 1) q <= 8'b0;
    else q <= d;
  end
endmodule
```
After formatting for three times, it generates three empty lines:
```
module top_module (
    input clk,
    input areset,
    input [7:0] d,
    output reg [7:0] q // Here is a comment

);
  always @(posedge clk or posedge areset) begin
    if (areset == 1) q <= 8'b0;
    else q <= d;
  end
endmodule
```

### When an error occurs

Before formatting:
```
module top_module (
    input clk,
    input areset,
    input [7:0] d,
    output reg [7:0] q,// Here is an extra ','
);
  always @(posedge clk or posedge areset) begin
    if (areset == 1) q <= 8'b0;
    else q <= d;
  end
endmodule
```
After formatting:
```
module top_module (

    input clk,

    input areset,

    input [7:0] d,

    output reg [7:0] q,// Here is an extra ','

);

  always @(posedge clk or posedge areset) begin

    if (areset == 1) q <= 8'b0;

    else q <= d;

  end

endmodule

```

## Expected Result

Of course an empty line should not be generated after a comment.

And when an error occurs, I prefer the formatter to do nothing instead of making empty lines.

baowuji, 2025-05-21:
I encountered the same problem.

bmpenuelas, 2025-05-22:
Thanks for opening the issue! Does this happen when formatting a selection, when formatting the whole file, or both?

CatalpaEel, 2025-05-23:
When formatting the whole file. ( Maybe the both, but I don't test it when formatting a selection. )

bmpenuelas, 2025-05-30:
Thanks to @cylindrical2002 for the excellent steps to reproduce the issue, I've just pushed a patch with fixes for both of the issues that you reported, and it should be live in the marketplace by now (3.3.0).

Thanks again for reporting!