bmpenuelas/systemverilog-formatter-vscode #30: Generates endless newlines in block comment after formatting

wild-duck, 2025-06-14:
Thanks for your formatter, which has provided me with a lot of convenience. However, it seems to have a bug after update.

I'm used to adding a block comment at the beginning of my files to note some file information. However, after updating, every time I format my code, it inserts two new blank lines between each line of the block comment. This causes my files to fill with a large number of useless blank lines. I've tried rolling back to older versions, and this issue only disappears when reverting to version 2.16.4 or earlier. It appears that the problem mentioned in [#28](https://github.com/bmpenuelas/systemverilog-formatter-vscode/issues/28) hasn't been completely resolved. 

# Example

## Before  formatting
```
/*
 ***************************************************************************************
 * @file: 
 * @author: 
 * @date: 
 * @function: 
 * @note:   
 * @version: 
 ***************************************************************************************
 */
```

## After formatting for one time
```
/*

 ***************************************************************************************

 * @file: 

 * @author: 

 * @date: 

 * @function: 

 * @note:   

 * @version: v1.0

 ***************************************************************************************

 */
```

## After formatting for two times
```
/*

 ***************************************************************************************

 * @file: 

 * @author: 

 * @date: 

 * @function: 

 * @note:   

 * @version: v1.0

 ***************************************************************************************

 */
```
I find this truly regrettable and hope this issue can be resolved soon.

bmpenuelas, 2025-06-14:
Oh I'll definitely try to fix this ASAP. Does this only happen with block comments? If you format a file without block comments does it work fine?

wild-duck, 2025-06-21:
> It does indeed format the block comment incorrectly for me as well, but not regular comments in the ports declaration. This is after formatting once, with the latest version:它确实也确实为我错误地格式化了块注释，但不是 ports 声明中的常规注释。这是在使用最新版本格式化一次之后：
> 
> [image]
> 
> Can you please try your block comment on top of this file, with the latest version? [test/example.sv](https://github.com/bmpenuelas/systemverilog-formatter-vscode/tree/master/test/example.sv)您能否尝试使用此文件的最新版本的块注释？test/example.sv First, I would like to confirm whether we get different behaviors on different machines or not.首先，我想确认一下我们是否在不同的机器上获得了不同的行为。
> 
> Also, if I change the file mode to LF it formats it correctly, so it's still something with the automatic conversion to CRLF that VSCode does.此外，如果我将文件模式更改为 LF，它会正确格式化它，因此它仍然是 VSCode 自动转换为 CRLF 的功能。

Yes, the results on my computer match what's shown in your screenshot. Furthermore, based on my testing, this behavior only occurs with block comments (regardless of their position in the file). When I remove the block comments, everything appears to function normally.

wild-duck, 2025-06-21:
[image]

Additionally, I observed in UEStudio's hex mode that the line breaks in my original file appear to be CRLF.

bmpenuelas, 2025-06-29:
Happy to help! I've made it into an official release now.

closed 2025-06-29 by bmpenuelas